metachris/RPIO #53: SystemError: This module can only be run on a Raspberry Pi!

momofoto, 2015-02-21:
Hi, 
I have a Raspberry Pi 2 and am getting the error below when trying to import the RPIO module. Should. Was getting this error when trying to import RPi.GPIO as well until recently, but that imports OK after some recent updates? Only just worked out that these are two different things - being a newbie and all that!

@raspberrypi ~ $ python
Python 2.7.3 (default, Mar 18 2014, 05:13:23) 
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/RPIO-0.10.1-py2.7-linux-armv7l.egg/RPIO/__init__.py", line 115, in <module>
    import RPIO._GPIO as _GPIO
SystemError: This module can only be run on a Raspberry Pi!
>>>

felix133711, 2015-02-26:
Hi,

same problem :(

netelbcn, 2015-03-08:
Hi!

Could you solve this error...? I'm now trying to import RPi.GPIO with 0.5.11.egg version, and the same error...

tylerwowen, 2015-03-25:
Hi,

This is issues is caused by the hardware changes in Pi 2. Some hardcoded parts are not working on Pi2. I forked it and fixed the problems. Now it is running on my Pi2, but it is not fully tested so I cannot guarantee its quality. So you can clone from my repo and install it. https://github.com/tylerwowen/RPIO

netelbcn, 2015-03-25:
Thanks!!

I finally solved this error deleting all RPi.GPIO files in the system and then installing from scratch the 0.5.11 version.

Anyway, thank you very much!

HasiKP, 2015-03-30:
@netelbcn  how do I delete RPI.GPIO files in the system and install 0.5.11 version from scratch

netelbcn, 2015-03-30:
@HasiKP:
Uninstall the version that dpkg/apt-get knows about.
sudo apt-get purge python{,3}-rpi.gpio

Scan your filesystem for all copies of RPi.GPIO
find /usr -name "*RPi*"
remove everything that you find

sudo apt-get install python{,3}-rpi.gpio
to put back a good version.

netelbcn, 2015-03-30:
@HasiKP:
Uninstall the version that dpkg/apt-get knows about.
`<addr>`sudo apt-get purge python{,3}-rpi.gpio

Scan your filesystem for all copies of RPi.GPIO
`<addr>`find /usr -name "*RPi*"
remove everything that you find

`<addr>`sudo apt-get install python{,3}-rpi.gpio
to put back a good version.

netelbcn, 2015-03-30:
@HasiKP:
Uninstall the version that dpkg/apt-get knows about.
`sudo apt-get purge python{,3}-rpi.gpio`

Scan your filesystem for all copies of RPi.GPIO
`find /usr -name "*RPi*"`
remove everything that you find

`sudo apt-get install python{,3}-rpi.gpio`
to put back a good version.

yah86, 2015-04-02:
@tylerwowen 

Hi,

I tried using your RPIO library, but the servo methods and PWM methods did not work.

tylerwowen, 2015-04-02:
@yah86 
Sorry for the inconvenience. I didn't modify the part for PWM. I'm busy recently, so I don't think I can fix till next week. You can try to modify files under source/c_pwm to make it work. Basically you just need to change some hardware addresses.

yah86, 2015-04-04:
@tylerwowen 

Thanks.
I'll try it.

HasiKP, 2015-04-05:
@netelbcn 
thanx
@yah86 
did you find a way to use servo methods and the pwm methods

yah86, 2015-04-06:
@HasiKP 
I have not yet found its way.
Because I have not yet found the datasheet of BCM2836.

joan2937, 2015-04-06:
The Pi2 peripherals are exactly the same as the earlier Pis.  See the bcm3835 documentation.  The peripheral base address has moved from 0x20000000 to 0x3F000000.  DMA memory is now at offset 0xC0000000 rather than 0x40000000.  For RPIO.GPIO you should be allocating DMA memory via the mailbox interface.

HasiKP, 2015-04-06:
@joan2937 
I'm new to this so its a great help for everyone if you can modify it and share with us.

joan2937, 2015-04-06:
I'm busy with other software and don't actually use RPIO.GPIO.  For examples of the needed changes have a look at servoblaster, pi-blaster, or pigpio (they each use PWM/PCM and DMA to generate PWM and servo pulses).

ankur5294, 2015-05-01:
@tylerwowen 
Have you fixed RPIO PWM and other bugs? Please let know asap!!!

r-shayestehpour, 2015-05-12:
any updates on this issue? 
(RPi 2, GPIO 0.5.11)

Vijayenthiran, 2015-08-05:
Any update on the issue? Having the same problem :(

griffegg, 2015-08-14:
Hi, me too. I'm building a robot and without RPIO working I have too much jitter. Do you have a working version for the RPi2 yet? Thanks so much for your efforts!

loganhe0215, 2015-08-15:
same problem

joan2937, 2015-08-15:
There are quite a few alternatives to RPIO.GPIO.  For instance, if all you want is servo pulses look at [servoblaster](https://github.com/richardghirst/PiBits/tree/master/ServoBlaster).  If you want a general Python gpio module including PWM/servo pulses plus much more look at my pigpio  [here](http://abyz.co.uk/rpi/pigpio/) and [here](https://github.com/joan2937/pigpio/tree/master).

Vijayenthiran, 2015-08-15:
@joan2937  Hi, I tried to use servoblaster in Rpi - 2 Model B. Didn't work. Even pi-blaster didn't work. Looks like DMA controller address for Rpi-2 is different from the previous model. Should try your pigpio library. !

joan2937, 2015-08-15:
@Vijayenthiran servoblaster and pi-blaster have been updated to work for the Pi2.  The servoblaster github repository has not been updated.  You can find a link to the updated software by searching for servoblaster on the rasperrypi.org forums.

griffegg, 2015-08-16:
Thank you so much! I will try them.

Vijayenthiran, 2015-08-16:
Thank you Joan.. Will try them.

rdagger, 2015-08-18:
Joan, the RPIO.add_channel_pulse(dma_channel, gpio, start, width) method has a start argument that allows you to specify when pulses start during the pwm period.  Does pigpio offer this functionality?  If not, is it something you would consider adding?  Thanks for the great library.

joan2937, 2015-08-18:
Probably best to raise an issue at https://github.com/joan2937/pigpio/tree/master unless it relates directly to RPIO.

slowmed, 2015-09-05:
no update for PWM yet ??

alexellis, 2015-12-18:
I spoke to Eben Upton at PiWars he said that there is a setting for /boot/config.txt which uses the memory addressing from the original PI. This should fix issues for PI2 but at the same time halves your RAM.

rdagger, 2015-12-19:
@alexellis I'm skeptical that it could work.  I took a quick look at the library several months ago and the pwm issues were more than just changes in address locations.  I don't recall specifically but I remember something about the need for a rewrite using the mailbox interface.  Still let us know if you have any luck.

psssix, 2016-01-04:
Good afternoon. I, too, was faced with this problem. I looked at forks RPIO library, and found the right fork. Good afternoon. I propose to use the current fork:
https://github.com/tylerwowen/RPIO

DonRecardo, 2016-03-10:
I take it thats its because I have a Pi 3 that I also get the msg , only runs on a raspberry Pi

ludusrusso, 2016-03-21:
I face the same problem with RasPi3, any solution yet?

limuxy, 2016-03-25:
@DonRecardo @ludusrusso I've forked @tylerwowen 's fork and added support for Pi 3, working for my Pi 3 now, get it at https://github.com/limuxy/RPIO
do let me know if you meet any problem

DonRecardo, 2016-03-25:
@limuxy   Firstly thanks for letting me know you have it sorted  which is very good of you .
 I dont mean to be a bigger nuisance but being a noob to both linux and R pi I now have another problem . Everything else I have installed I have done with sudo apt-get install , but this package is a zip  and I dont really know how to handle it .
I downloaded it and it appears in my download folder but if I double click on it it asks me what program I wish to you to unzip it  and I dont know what to use or where to find it , Then when I do get it unzipped 
how do I install it ?  I can only assume its something like sudo python setup.py ?
I could appreciate some help as what to do 

Cheers 
Don

DonRecardo, 2016-03-25:
@limuxy 
I spoke to soon . It wasnt my attempt at an installation that was running . It was trying to run the original install at /usr/local/lib/python3.4/dist-packages and of course thats the one that comes back with 
can only be run on a raspberry pi 

I have now removed the old version , but still dont know how or where to install your fork of it

DonRecardo, 2016-03-25:
@limuxy 
Well at the risk of being wrong again I will stick my neck out again and say I think its working 

After uninstalling the old version of RPIO I again did sudo python setup.py install
and it did its stuff
I then went and had a look and in usr/local/lib/python2.7/dist-packages and RPIO was there
but when I looked in python3.4 it didnt have it in its dist-packages  
As I use python3 I thought I must have to get it installed there too so I tried 
the same as before but substituted python with python3 as
sudo python3 setup.py install . and I thought I had won as it started its stuff but at the end it said it had errored  and on checking python3s dist -packeges it still wasnt there .
but I thought I would give it a try 
I stareted python3 up and did 
import RPIO ; print(dir(RPIO))
and it gave me  __doc__ , __loader__  ........etc
before all I got was "can only run on a raspberry pi"
so it appears to be working  ( I think )

Don

sanktnelson, 2016-05-12:
Hi all, I came here with the same problem. tried both versions from @tylerwoven and @limuxy and neither work on my raspberry 2. They both load fine, but add_channel_pulse does not seem to do anything, at least no change on the GPIO pin is visible. Any clues?

VSkublin, 2017-04-10:
I have faced the same problem, looked for solution and this is the solution, that make it works:
First need to be sure that header files and static libraries for python dev. installed  properly:
sudo apt-get install python-dev  # for python2.x installs
sudo apt-get install python3-dev  # for python3.x installs
Then try to install RPIO branch v2, it will work with RPI3.
cd ~
git clone https://github.com/metachris/RPIO.git --branch v2 --single-branch
cd RPIO
sudo python setup.py install

This solution helped me install RPIO for several RPI3s.

letroll, 2018-08-15:
same problem on archlinux arm 64bits because cpuinfo have no Hardware part in it's return so the system can't figure it's a raspberry pi

burnzzz, 2019-01-28:
Exactly. Archlinux on Raspberry 3B+ not working.

bdn76, 2019-03-06:
Same problem on RPi 3B+ with Ubuntu 18.04.2 64bit.

captnjohnny1618, 2019-06-30:
Also running into this issue with 0.6.5 (PyPI) module on 3B with Ubuntu 18.04.2 64bit.  Problem is fixed in 0.6.6a but a push to PyPI soon would be awesome!

pierrotsmnrd, 2020-09-27:
I had installed python3-rpi.gpio with pip. I realized the version was wrong. I uninstalled it, downloaded from the source ( https://sourceforge.net/p/raspberry-gpio-python/code/ci/default/tree/RPi/ ) and built and installed manually (python3 setup.py build && python3 setup.py install), and now it works - even on my Raspberry Pi 4 .

The last version has a better handling of the last models of raspberry pis and their revision codes ( https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md ).