metachris/RPIO #54: RPIO install error

peter-mason-dpm, 2015-02-22:
I did this on my RPi2
$ sudo apt-get install python-setuptools
$ sudo easy_install -U RPIO

and got this

pi@raspberrypi ~/RPIO $ sudo easy_install -U RPIO
Searching for RPIO
Reading http://pypi.python.org/simple/RPIO/
Best match: RPIO 0.10.0
Downloading https://pypi.python.org/packages/source/R/RPIO/RPIO-0.10.0.tar.gz#md5=cefc45422833dcafcd59b78dffc540f4
Processing RPIO-0.10.0.tar.gz
Running RPIO-0.10.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-iyUVZN/RPIO-0.10.0/egg-dist-tmp-mlGrcH
source/c_gpio/py_gpio.c:28:20: fatal error: Python.h: No such file or directory
compilation terminated.
error: Setup script exited with error: command 'gcc' failed with exit status 1

I tried this 

$ git clone https://github.com/metachris/RPIO.git
$ cd RPIO
$ sudo python setup.py install

with a similar result.
can anyone tell me what I need to do please?

joan2937, 2015-02-24:
It looks like you haven't got the Python development environment.

sudo apt-get install python-dev

and/or

sudo apt-get install python3-dev

peter-mason-dpm, 2015-02-24:
Joan
I started with unix last century but did more harware than software and finding what I need to do now, then how to do it is a mystery.At the dawn of the epoch with assmbler it was all more obvious.

thank you.
installing python3-dev did the trick and GPIO seems to have installed, BUT python still does not import it.

joan2937, 2015-02-24:
I've just installed for python 2 and 3 and both work. You have to launch Python with sudo to be able to import the module.  It does not work on the new model Pi 2.

peter-mason-dpm, 2015-02-24:
I have RPi2. I would like a good pwm module.

joan2937, 2015-02-24:
My pigpio library does PWM (among other things).  http://abyz.co.uk/rpi/pigpio/

peter-mason-dpm, 2015-02-24:
The scope looks interesting I will look at that with interest.
Does the RPi2 do C++?

joan2937, 2015-02-24:
Yes, all the normal packages are available on the Pi 2.  gcc supports C/C++ (and the rest) out of the box.

closed 2015-02-25 by peter-mason-dpm

cuchiaram, 2017-08-16:
Wow closed back in 2015....I must be goofy because this should now be seamless but  I've tried all online recommendations to get my Pi 3B with Raspian to recognize RPIO in any way form or fashion. hmmmm