metachris/RPIO #67: RPIO 0.10.0 did not work on RaspberryPi V2

sebathorus, 2015-06-10:
Hello

I tried to use your library to generate some PWM signals on a RaspberryPi V2 but I get always same error:
Traceback (most recent call last):
  File "/usr/local/bin/rpio", line 5, in <module>
    pkg_resources.run_script('RPIO==0.10.1', 'rpio')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 499, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1235, in run_script
    execfile(script_filename, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/RPIO-0.10.1-py2.7-linux-armv7l.egg/EGG-INFO/scripts/rpio", line 353, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/RPIO-0.10.1-py2.7-linux-armv7l.egg/EGG-INFO/scripts/rpio", line 147, in main
    import RPIO
  File "/usr/local/lib/python2.7/dist-packages/RPIO-0.10.1-py2.7-linux-armv7l.egg/RPIO/__init__.py", line 115, in <module>
    import RPIO._GPIO as _GPIO
SystemError: This module can only be run on a Raspberry Pi!

Any help on solving this would be welcome.
Thank you
 -=Seba=-

K1773R, 2015-06-10:
check the current pull requests

sebathorus, 2015-06-11:
Ok, thank you.

rburkhal, 2015-06-19:
Using the updated version from your fork, I don't get any PWM activity.  Still new to the hardware, let me know if I can help, I do have 2 Pi2's, and some skills that could assist.

MakiseKurisu, 2016-04-12:
Any update on this issue?