metachris/RPIO #74: set_servo is momentary not permanent

Berryfier, 2015-12-23:
A test using the lines:
    from RPIO import PWM
    servo = PWM.Servo()
    servo.set_servo(04, 1500)
sends the servo on its way, but has to be run several times for the servo to reach its destination. Running the program lines causes PWM pulses to be sent to the servo only for a short instant, after which the servo can again be freely moved by hand. On running the program lines RPIO reports:
    Using hardware: PWM
    PW increments:  10us
    Initializing channel 0...
    add_channel_pulse: channel=0, gpio=4, start=0, width=150
    init_gpio 4
    shutting down dma channel 0
    clear_channel: channel=0
I am using Pi B+ and would like the servo control pulses to be sent continuously until deliberately stopped.

K1773R, 2015-12-23:
they are all non blocking calls, got to create a loop

Berryfier, 2015-12-24:
@K1773R . Thanks for the diagnosis and prgnosis. By adding time.sleep(2) after servo.set_servo I find pulses are sent to the servo until time.sleep runs out.

closed 2015-12-24 by Berryfier