metachris/RPIO #77: Is this project dead?

DJSymBiotiX, 2016-02-24:
It's been a while since I've done Pi development, but it seems to me like this lib is missing code for the zero and the + series? I can't seem to use any of the extended header pins with this library on my zero, and trying to get the info just returns with "KeyError: '900092'"

Just curious if this is still actually being developed, or if I should be looking elsewhere?

chdsbd, 2016-03-04:
I gave up on using this project and transferred over to using [RPi.GPIO](https://pypi.python.org/pypi/RPi.GPIO), which provides pretty much the same functionality.

DJSymBiotiX, 2016-03-04:
Yeah that seems to be what I was thinking of doing. It's a shame since this lib has a few things that I liked.

karel-kahula, 2016-03-04:
ya me 2

alexellis, 2016-03-09:
If this was fixed it would serve a lot of people with PI2s wanting to generate PWM signals - it was perfect for robots (servos, motor drivers, LEDs).

As far as I know it still works on ARMv6 - PI A/A+/B/B+/Zero. An alternative may to be to use a tiny 2 dollar Arduino nano as a co-op board over USB with something like PyFirmata

chdsbd, 2016-03-09:
@alexellis [RPi.GPIO](https://pypi.python.org/pypi/RPi.GPIO) provides basically all of the functionality of this library (PWM included) while also being maintained and compatible with the latest Raspberry Pis.

DJSymBiotiX, 2016-03-09:
Not necessarily. One example I can think of is that this lib allows you to set up an interrupt on any edge, and the function it calls actually passes in which edge the interrupt fired on, while in RPi.GPIO you can only choose up, down, or both, but on both it doesn't tell you which it triggered on.

alexellis, 2016-03-09:
@chdsbd RPi.GPIO really does not match this library's key feature - which I would say is **hardware PWM signals through DMA**. Even the front page of "RPi.GPIO" states how this is only covered through software - so much less reliable and prone to jitter / higher CPU loads.

If PWM is not important to your projects, then one of the other libraries should be fine for your purposes.

chdsbd, 2016-03-09:
@alexellis I have been using RPi.GPIO to trigger servos without issue, so software PWM is at least reliable enough to control servos. Maybe there would be issues controlling LEDs precisely.

DEvil0000, 2016-03-10:
RPi.GPIO is fine for simple things but for my use cases it is much to slow.
I really need the DMA stuff so I will stick with this and fix what I need.

@chdsbd It is just not true to say RPi.GPIO provides the same features or performance.

rdagger, 2016-03-10:
Agree that this library provides some useful features not found in others.  I looked into fixing it a few months ago and it was not trivial.  If I recall correctly, it wasn't a matter of just updating addresses.  Fundamental changes related to [Mailbox property interfaces](https://github.com/raspberrypi/firmware/wiki/Mailbox-property-interface) were needed.

lurch, 2016-05-15:
Even _without_ adding PWM support for the Pi2 / Pi3, it would be great if the version of RPIO on pypy could be updated to support B+ / A+ / PiZero 'out of the box' without having to manually patch the code.

lurch, 2016-05-15:
For anybody looking for accurate PWM, there's also the http://abyz.co.uk/rpi/pigpio/ project.

JennaSys, 2016-05-22:
RPIO is a great clean pythonic library, but without it being maintained for the newer versions of the RasPi it doesn't do much good.  I've since moved on to using pigpio which is very full featured.  That one requires you to run a daemon on the RasPi, but doing so also lets you do remote calls, making it possible to develop code on another dev machine (like a laptop) while still allowing your code to access the GPIO port on a RasPi.

metachris, 2016-06-02:
RPIO has been reactivated and support for Pi 0, 2 and 3 is coming: https://github.com/metachris/RPIO/tree/v2 

If everything works, an update is coming to PyPi end of next week.

Thanks for the links and discussion here!

DJSymBiotiX, 2016-06-02:
Thats great to hear. Thanks for chiming in!

rdagger, 2016-06-02:
Very cool.  Look forward to trying it.
Thanks.

lurch, 2016-06-02:
That's great news @metachris :-)
As you may or may not be aware, [GpioZero](https://github.com/RPi-Distro/python-gpiozero) also provides an (optional) [wrapper](http://gpiozero.readthedocs.io/en/latest/api_pins.html) for RPIO, and the initial plans for the next v1.3 release of GpioZero are now being formed... https://github.com/RPi-Distro/python-gpiozero/issues/350

metachris, 2016-06-02:
Thank you for the information! I was not aware that GpioZero has an optional wrapper for RPIO. I'm definitely open for ideas on how to move RPIO forward.

rdagger, 2016-06-05:
Just tested 2.0.0-beta1 on Raspberry Pi 3 and PWM is working.  Great job.  Thanks.

lurch, 2016-09-01:
> If everything works, an update is coming to PyPi end of next week.

@metachris it looks like this project has stalled again? :-(

TafThorne, 2017-07-31:
The testing here looks like it would help verify #82.  

@lurch The current maintainer (@metachris) has said that he does not have a lot of time to give to RPIO at the moment.  He seems happy to give more people write access to the project if they would like to help with keeping it active.  See pull request #82 for details.