metachris/RPIO #78: Don't work on Raspberry Pi 3

chdorb, 2016-03-13:
When importing on python shell or launching rpio/rpio-ncurses, there is an error:
```
pi@raspberrypi:~ $ sudo rpio
Traceback (most recent call last):
  File "/usr/local/bin/rpio", line 345, in <module>
    main()
  File "/usr/local/bin/rpio", line 147, in main
    import RPIO
  File "/usr/local/lib/python2.7/dist-packages/RPIO/__init__.py", line 115, in <module>
    import RPIO._GPIO as _GPIO
SystemError: This module can only be run on a Raspberry Pi!
```

That's on Raspberry Pi 3.

zacharylawrence, 2016-03-25:
I'm having the same problem on my Raspberry Pi 3. Looks like it might have happened in the past with the Raspberry Pi 2 due to hardware changes: https://github.com/metachris/RPIO/issues/53. Any ideas on how to get this working?

limuxy, 2016-03-26:
@zacharylawrence tylerwowen's fork here: https://github.com/tylerwowen/RPIO works for Pi 2 and I've added support for Pi 3 basing on his fork, get mine here: https://github.com/limuxy/RPIO
so far it works on my Pi 3

ndphu, 2016-04-15:
@limuxy 
I installed by cloning your repo. Everything is ok. I can import RPIO, and the 'SystemError' message is gone.

But when using the lib, it's not working.

Here is my sample script:

`import  time
from  RPIO import  PWM
servo =  PWM.Servo ()
servo.set_servo ( 17 , 900 )
try :
        while  True :
                servo.set_servo ( 17 , 750 )
                time.sleep ( 1 )
                servo.set_servo ( 17 , 1200 )
                time.sleep ( 1 )
                servo.set_servo ( 17 , 2500 )
                time.sleep ( 1 )
except KeyboardInterrupt:
        servo.stop_servo ( 17 )
`

I tested the servo and GPIO pin 17 still working with the software PWM (GPIO.PWM). Please take a look if you have time. 

Thanks

ndphu, 2016-04-15:
The same code worked on Pi Zero. I installed RPIO on my Pi Zero the same way as I did with the Raspberry Pi 3.

dalanmiller, 2016-04-27:
@limuxy How does one install your version of RPIO? I can't figure out how to build the .egg or install

limuxy, 2016-04-28:
@ndphu codes concerning PWM are not touched so it's not working on both tylerwowen's and mine fork.... only the GPIO part works, maybe later someone will look into the PWM part and make it work on Pi 2/3

@dalanmiller `python setup.py install` will do it, you'll need `apt-get install python-dev` if it says 'python.h not found'

ndphu, 2016-04-28:
@limuxy I'm now switch to use servoblaster as an alternative. My purpose is for controlling servo rotation so it fit my need.

alex1893, 2016-05-14:
Hello , 
I´m new here and I have this problem as you .
I´m a begginer in linux 
pi@raspberrypi:~/Downloads/proiect $ sudo python capture-positives.py
Traceback (most recent call last):
  File "capture-positives.py", line 14, in <module>
    import hardware
  File "/home/pi/Downloads/proiect/hardware.py", line 8, in <module>
    import RPIO
  File "build/bdist.linux-armv7l/egg/RPIO/__init__.py", line 115, in <module>
  File "build/bdist.linux-armv7l/egg/RPIO/_GPIO.py", line 7, in <module>
  File "build/bdist.linux-armv7l/egg/RPIO/_GPIO.py", line 6, in __bootstrap__
SystemError: This module can only be run on a Raspberry Pi!

Can someone help me step by step please ?
Thanks

alex1893, 2016-05-15:
Some help please?
I know the solution is here but I don´t know how to apply it 

Thanks

metachris, 2016-06-02:
Changes are in https://github.com/metachris/RPIO/tree/v2 

If everything works, an update is coming to PyPi end of next week.

dalanmiller, 2016-06-16:
Any updates on this?

rdagger, 2016-06-17:
@dalanmiller It worked for me on Raspberry Pi 3.  I only tested the PWM.  Here is how I installed:

> cd ~
> git clone https://github.com/metachris/RPIO.git --branch v2 --single-branch
> cd RPIO
> sudo python setup.py install

mjlandry27, 2016-06-29:
All-

Running a Pi 3 with Raspbian Jessie. Installed RPIO branch v2 as described by rdagger above.

rpio-curses returns the following:

`
$ rpio-curses
Script not started as root. Running sudo...
Traceback (most recent call last):
  File "/usr/local/bin/rpio-curses", line 4, in <module>
    __import__('pkg_resources').run_script('RPIO==2.0.0-beta1', 'rpio-curses')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 534, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/RPIO-2.0.0_beta1-py2.7-linux-armv7l.egg/EGG-INFO/scripts/rpio-curses", line 96, in <module>
    
  File "/usr/local/lib/python2.7/dist-packages/RPIO-2.0.0_beta1-py2.7-linux-armv7l.egg/EGG-INFO/scripts/rpio-curses", line 78, in get_gpiolist
    
  File "build/bdist.linux-armv7l/egg/RPIO/__init__.py", line 186, in sysinfo
KeyError: 'a02082'
`

Any advice? Thanks in advance for your help with this!

doom-fr, 2016-07-18:
@metachris : using installation of branch v2 on raspberry pi 3 as described by @rdagger.
I have the same problem as @mjlandry27 (even if I have used the `sudo` prefix).
`KeyError: 'a02082'`
as well for `sudo rpi-curses`, for `sudo rpio --sysinfos` and also for `sudo rpio -i 17`.
Tried also with python3 `sudo python3 setup.py install`. Problem is the same.

value `a02082` seems to be the revision field supplied by `cat /proc/cpuinfo` if this could help you to solve this issue.

thx

doom-fr, 2016-07-18:
Yep,

I have added one line in the MODEL_DATA table : `'a02082' : ('B', '3.0', 1024, '?')`
Don't know if it is the good solution but `sudo rpio --sysinfo` works !

But when I do `sudo rpio -i 17` there a pb in rpio script line 233 in main `RPIO._RPIO.GPIO_FUNCTION[f], \` with error `KeyError : 2`.
It seems that in _RPIO.py, the `GPIO_FUNCTIONS = {0: "OUTPUT", 1: "INPUT", 4: "ALT0", 6:"ALT2", 7: "-"}` does not know the 2 key.
if I type `sudo rpio -i 3`, it works,but
with `sudo rpio -i 4`, there's a keyError 5 !

`rpio-curses` don't wan't to launch for a similar problem.

Not sure what should be done.

D.

doom-fr, 2016-07-27:
Have fixed it : https://github.com/doom-fr/RPIO/tree/v2b2
I have added tests on keys for GPIO_FUNCTIONS and MODEL_DATA dictionnaries before getting values.
Think it will fix other incompatibilities. Not fully tested.
Hope it will have no border effects !

Have fun !
_(and be cool : it's my first source contribution)_

sw-yx, 2016-07-31:
@doom-fr : i tried your fork (https://github.com/doom-fr/RPIO) and it doesnt seem to work on RPi3's? I am getting the "can only be run on a Raspberry Pi" error :(

@rdagger @mjlandry27 : I tried the v2 from @metachris and it ran well but the pwm.servo method still doesnt seem to be working. can you confirm if it is working for you? i have no idea if its because i have a defective servo

@limuxy if you ever get around to fixing the pwm/servo part of this, I would gladly patreon you some money...

doom-fr, 2016-07-31:
@sw-yx : cool. To se the problem, can you tell me your Hardware and revision value in /proc/cpuinfo
Me I have :
`Hardware	: BCM2709
Revision	: a02082`

nargetdev, 2017-02-17:
@limuxy branch worked for me, thanks!!

VSkublin, 2017-04-10:
I have faced the same problem, this solution make it work:
First need to be sure that header files and static libraries for python dev. installed  properly:
sudo apt-get install python-dev  # for python2.x installs
sudo apt-get install python3-dev  # for python3.x installs
Then try to install RPIO branch v2, it will work with RPI3.
cd ~
git clone https://github.com/metachris/RPIO.git --branch v2 --single-branch
cd RPIO
sudo python setup.py install

This solution helped me install RPIO for several RPI3s.

jpg35310, 2017-05-11:
Hello,

For information it works very well on a RASPBERRY PI A + with 3 servos PWM.
Many thanks to all the people who participated in this project.

Regards,

Dave4675, 2017-08-30:
Hi to all, I tryed to install the branch 2 with the command git clone https://github.com/metachris/RPIO.git --branch v2 --single-branch, then I went into the folder and then installed this version. Now I have a raspberry pi 0 w and it doesn't work for me, it say after typed sudo rpio-curses:

Traceback (most recent call last):
  File "/usr/local/bin/rpio-curses", line 4, in <module>
    __import__('pkg_resources').run_script('RPIO==2.0.0-beta1', 'rpio-curses')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 534, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/RPIO-2.0.0_beta1-py2.7-linux-armv6l.egg/EGG-INFO/scripts/rpio-curses", line 96, in <module>

  File "/usr/local/lib/python2.7/dist-packages/RPIO-2.0.0_beta1-py2.7-linux-armv6l.egg/EGG-INFO/scripts/rpio-curses", line 78, in get_gpiolist

  File "build/bdist.linux-armv6l/egg/RPIO/__init__.py", line 186, in sysinfo
KeyError: '9000c1'

Can you help me to fix this?

My cpu info are:
Hardware      : BCM2835
Revision        : 9000c1

changtimwu, 2017-10-27:
Hi! 
My Raspbian 9 on PI3  also suffers such a problem
``` 
cat /proc/cpuinfo
Hardware        : BCM2835
Revision        : a22082
```

I found the procedure of clone/build/install the `v2` branch can be simplified as one single command. 
``` 
sudo pip install https://github.com/metachris/RPIO/archive/v2.zip
```

dazza2017, 2017-11-29:
Greetings to All and Metachrist,
I have tried every one of these attempts at fixing the "This module can only be run on a Raspberry PI!" problem, but no joy. I am at my wits end. I am either going to tear my hair out or kill myself ( which ever is the least painful) if I don't find a solution !!
please hellllp . its October 29th 2017

rdagger, 2017-11-30:
@dazza2017 I just tried installing the RPIO library on a Pi 3 running the latest version of Raspbian stretch and the v2 installs but it is not stable.  PWM caused the Pi to freeze.  I recommend you try the [PiGPIO library](http://abyz.me.uk/rpi/pigpio/) which comes installed with the latest version of Raspbian.  The author Joan is very active on the Raspberry Pi [forum](https://www.raspberrypi.org/forums/) and she is quick to answer questions and address problems.

dslimmer, 2018-06-08:
I was able to get rpio and rpio-curses working from the  RPIO v2 branch (RPIO-2) on a Pi 3 B+ with the following source code modifications.

In RPIO-2/source/RPIO/`__init__.py`, line 162   (at the end of MODEL_DATA)   I added     
     ` 'a020d3': ('B+', '1.3', 1024, 'Sony')`

In RPIO-2/source/c_gpio/py_gpio.c, line 99   changed from   
 `return *(*gpio_to_pin+bcm_gpio_id);`     to     `return (int)*(gpio_to_pin+bcm_gpio_id);`

In RPIO-2/source/c_gpio/py_gpio.c, line 108  also looks like it should be changed from    
`return *(*pin_to_gpio+board_pin_id);`     to     `return (int)*(pin_to_gpio+board_pin_id);` 

In RPIO-2/source/c_gpio/c_gpio.c, line 35  changed from      
   `#define BCM2708_PERI_BASE   0x20000000`        to      `#define BCM2708_PERI_BASE   0x3F000000`

After making these changes redo the install:
`cd RPIO-2`
`sudo python setup.py install`

I was also able to get the RPIO-2/source/tests_gpio.py to run (with a few errors)  by renaming the directory RPIO-2/srouce/RPIO  to  RPIO-2/source/RPIOhide, and then:
`sudo python ./tests_gpio.py`

Don't forget to restore the original directory name RPIO-2/srouce/RPIO if you want to run the install again.

balblub56754, 2019-02-01:
When I try to compile the v2 branch on my voidliunx I got this compile error. Any guess what I could change to make it work?

Thanks for any help!

source/c_gpio/c_gpio.c:79:9: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
     if ((uint32_t)gpio_mem % PAGE_SIZE)
         ^
source/c_gpio/c_gpio.c:80:34: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
         gpio_mem += PAGE_SIZE - ((uint32_t)gpio_mem % PAGE_SIZE);
                                  ^
source/c_gpio/c_gpio.c:82:35: error: 'caddr_t' undeclared (first use in this function)
     gpio_map = (uint32_t *)mmap( (caddr_t)gpio_mem, BLOCK_SIZE, PROT_READ|PROT_WRITE, MAP_SHARED|MAP_FIXED, mem_fd, GPIO_BASE);
                                   ^~~~~~~
source/c_gpio/c_gpio.c:82:35: note: each undeclared identifier is reported only once for each function it appears in
source/c_gpio/c_gpio.c:82:43: error: expected ')' before 'gpio_mem'
     gpio_map = (uint32_t *)mmap( (caddr_t)gpio_mem, BLOCK_SIZE, PROT_READ|PROT_WRITE, MAP_SHARED|MAP_FIXED, mem_fd, GPIO_BASE);
                                           ^~~~~~~~
                                           )
source/c_gpio/c_gpio.c:82:28: error: too few arguments to function 'mmap'
     gpio_map = (uint32_t *)mmap( (caddr_t)gpio_mem, BLOCK_SIZE, PROT_READ|PROT_WRITE, MAP_SHARED|MAP_FIXED, mem_fd, GPIO_BASE);
                            ^~~~
In file included from source/c_gpio/c_gpio.c:32:
/usr/include/sys/mman.h:110:7: note: declared here
 void *mmap (void *, size_t, int, int, int, off_t);
       ^~~~
source/c_gpio/c_gpio.c:84:9: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
     if ((uint32_t)gpio_map < 0)
         ^
source/c_gpio/c_gpio.c: In function 'cleanup':
source/c_gpio/c_gpio.c:166:13: error: 'caddr_t' undeclared (first use in this function)
     munmap((caddr_t)gpio_map, BLOCK_SIZE);
             ^~~~~~~
source/c_gpio/c_gpio.c:166:21: error: expected ')' before 'gpio_map'
     munmap((caddr_t)gpio_map, BLOCK_SIZE);
                     ^~~~~~~~
                     )
source/c_gpio/c_gpio.c:166:5: error: too few arguments to function 'munmap'
     munmap((caddr_t)gpio_map, BLOCK_SIZE);
     ^~~~~~
In file included from source/c_gpio/c_gpio.c:32:
/usr/include/sys/mman.h:111:5: note: declared here
 int munmap (void *, size_t);
     ^~~~~~
error: command 'gcc' failed with exit status 1

GoogyGao, 2022-11-21:
> 

juses! You're the man!