metachris/RPIO #79: Cannot get RPIO.PWM to work with OSMC. Error: "Page 0 not present"

CieNTi, 2016-03-20:
Hello,

I just installed OSMC (Raspbian-Debian based) and tried to run a Pan-Tilt cam. The solution found depends on RPIO and it's PWM + set_servo.

### Short description

When first "set_servo" is called, program exits with:

```
RuntimeError: rpio-pwm: Page 0 not present (pfn 0xa100000000003c2e)
```

(pfn value changes every execution)

I tried both easy_install method (0.10.0) and your published .deb (0.10.1) and both fails.

### Long description

picam.py excerpt (from [MattStultz's github](https://github.com/MattStultz/PiCam.git))

```python
from RPIO import PWM
import time
import atexit
from flask import Flask, render_template, request
app = Flask(__name__)

# This function maps the angle we want to move the servo to, to the needed PWM value
def angleMap(angle):
        return int((round((1950.0/180.0),0)*angle) +550)

# Create a dictionary called pins to store the pin number, name, and angle
pins = {
    23 : {'name' : 'pan', 'angle' : 90},
    22 : {'name' : 'tilt', 'angle' : 90}
    }

# Create two servo objects using the RPIO PWM library
servoPan = PWM.Servo()
servoTilt = PWM.Servo()

# Setup the two servos and turn both to 90 degrees
servoPan.set_servo(23, angleMap(90))
```

This last line generates this error:
```
osmc@osmc:~/PiCam$ sudo python picam.py
Using hardware: PWM
PW increments:  10us
Initializing channel 0...
Traceback (most recent call last):
  File "picam.py", line 23, in <module>
    servoPan.set_servo(23, angleMap(90))
  File "/usr/lib/python2.7/dist-packages/RPIO/PWM/__init__.py", line 212, in set_servo
    init_channel(self._dma_channel, self._subcycle_time_us)
  File "/usr/lib/python2.7/dist-packages/RPIO/PWM/__init__.py", line 97, in init_channel
    return _PWM.init_channel(channel, subcycle_time_us)
RuntimeError: rpio-pwm: Page 0 not present (pfn 0xa100000000003c2e)
```

What is happening and what do you need from me to test?

Thanks a lot

alexellis, 2016-03-20:
The library is unmaintained by author. I'd suggest going elsewhere - an
Arduino nano can manage the servos very well over a USB cable.

CieNTi, 2016-03-20:
Ouch .. thanks a lot

DEvil0000, 2016-03-20:
Even if it is not maintained - the code is easy to understand. You can fix things yourself.
In the case of this issue i think it is releated to cpp/python or maybe os/hardware.
I think pfn means pointer to function.I have seen such a adress somewhere i think. May be one of the hardware mappings to access the gpio's.
Take a look at the cpp code and the specs.
But I am only guessing here. I have just read the code a while before. 
Hope this might help.

alexellis, 2016-03-21:
There is a thread open discussion the fixes needed, they are not easy. But
if you think they are then there are a bunch of us who would appreciate
your pull request.

DEvil0000, 2016-03-21:
OK it might not be that easy - let me try to help on this.

Your RPI is version 1 right?

I think about 2 things at the moment.
1. the python <-> cpp bridge is not working correct
 Can you try it using the C code?
 What python version is installed?
 32/64 Bit?
 How about github clone and compile?
2. the channel might not be usable somehow.
 Which channel are you trying to use?
 How about other channels?

And maybe a third thing:
3. only happening on OSMC? Or is this also happening on normal/vanilla Raspbian?
https://discourse.osmc.tv/t/hyperion-with-ws2812b-not-working-after-latest-updates/10844/19

closed 2021-08-21 by CieNTi