metachris/RPIO #80: Fails to run or Raspberry PI 3, gives error, "Can only run on Raspberry PI"

pnduffy, 2016-05-11:
Apparently the package does not recognize the PI3.

lurch, 2016-05-15:
Duplicate of #78

metachris, 2016-06-02:
Changes are in https://github.com/metachris/RPIO/tree/v2 

If everything works, an update is coming to PyPi end of next week.

JoBergs, 2016-07-27:
v2 works well on my RPi3 with the newest Jessie, thanks!

dalanmiller, 2016-08-02:
`v2` branch isn't working for me, I just cloned the repo switched to the branch and installed via `pip`:

```
sudo rpio-curses
Traceback (most recent call last):
  File "/usr/local/bin/rpio-curses", line 96, in <module>
    GPIO_LIST = get_gpiolist()
  File "/usr/local/bin/rpio-curses", line 78, in get_gpiolist
    if RPIO.sysinfo()[1] == 'B+':
  File "/usr/local/lib/python2.7/dist-packages/RPIO/__init__.py", line 186, in sysinfo
    return (RPI_REVISION_HEX,) + MODEL_DATA[RPI_REVISION_HEX.lstrip("0")]
KeyError: 'a22082'
```

JoBergs, 2016-08-02:
I didn't try rpio-curses but imported RPIO in Python.

lurch, 2016-08-02:
Yeah, looks like https://github.com/metachris/RPIO/blob/v2/source/RPIO/__init__.py#L150 (and various other parts) still need to be updated with the revision-codes for the Pi2, Pi3, PiZero, etc.

@metachris see https://github.com/metachris/RPIO/pull/82/files for an example of where parts need to be changed to cope with different revision-codes / model-numbers, etc.