metachris/RPIO #83: KivyPie only showing a Terminal!

RedChallenger4, 2016-06-25:
The first thing that popped up was a terminal asking for login and password.

So I entered the login and password. Yet, after it displayed "Welcome to KivyPie", it only showed a terminal below it (i.e. shows ' sysop@kivypie' and that's it)

How do I access the part about game/app development?

RedChallenger4, 2016-06-25:
By the way, I installed by reading the KivyPie image into SD card.

wgwz, 2016-06-26:
Wait, why is this in RPIO?

RedChallenger4, 2016-06-26:
I thought this was the Group page for problems with installing/using KivyPie?

wgwz, 2016-06-26:
No, this is the wrong place (I think). RPIO is a Python library for dealing with the GPIO pins on a Raspberry Pi. Nothing to do with Kivy (out of the box).

RedChallenger4, 2016-06-26:
I think the website wasn't updated because the help link lead me here. Is there an updated link other than StackExchange (tried that)?

K1773R, 2016-06-26:
no. this is the RPIO repository with a python module for GPIO handling.

wgwz, 2016-06-26:
@RedChallenger4 You should close the issue here. This is definitely not the right place.

closed 2016-06-26 by RedChallenger4