metachris/RPIO #85: Can't find python.h on PI3

waynejacobsen, 2016-08-10:
error code follows:
pi@raspberrypi:~/python/RPIO $ sudo python setup.py install
running install
running bdist_egg
running egg_info
creating source/RPIO.egg-info
writing source/RPIO.egg-info/PKG-INFO
writing top-level names to source/RPIO.egg-info/top_level.txt
writing dependency_links to source/RPIO.egg-info/dependency_links.txt
writing manifest file 'source/RPIO.egg-info/SOURCES.txt'
reading manifest file 'source/RPIO.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'source/RPIO.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/RPIO
copying source/RPIO/__init__.py -> build/lib.linux-armv7l-2.7/RPIO
copying source/RPIO/Exceptions.py -> build/lib.linux-armv7l-2.7/RPIO
copying source/RPIO/_RPIO.py -> build/lib.linux-armv7l-2.7/RPIO
creating build/lib.linux-armv7l-2.7/RPIO/PWM
copying source/RPIO/PWM/__init__.py -> build/lib.linux-armv7l-2.7/RPIO/PWM
running build_ext
building 'RPIO._GPIO' extension
creating build/temp.linux-armv7l-2.7
creating build/temp.linux-armv7l-2.7/source
creating build/temp.linux-armv7l-2.7/source/c_gpio
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c source/c_gpio/py_gpio.c -o build/temp.linux-armv7l-2.7/source/c_gpio/py_gpio.o -Wno-error=declaration-after-statement
source/c_gpio/py_gpio.c:28:20: fatal error: Python.h: No such file or directory
 #include "Python.h"
                    ^
compilation terminated.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

K1773R, 2016-08-10:
```sudo apt-get install python2.7-dev```

waynejacobsen, 2016-08-11:
Thats, it worked but now I get
pi@raspberrypi:~/python $ sudo python PIstuffing.py
Traceback (most recent call last):
  File "PIstuffing.py", line 3, in <module>
    from RPIO import PWM
  File "build/bdist.linux-armv7l/egg/RPIO/__init__.py", line 115, in <module>
  File "build/bdist.linux-armv7l/egg/RPIO/_GPIO.py", line 7, in <module>
  File "build/bdist.linux-armv7l/egg/RPIO/_GPIO.py", line 6, in __bootstrap__
SystemError: This module can only be run on a Raspberry Pi!

pradeephrish, 2017-01-03:
Switch branch to V2 and try again