metachris/RPIO #86: This module can only be run on a Raspberry Pi!

waynejacobsen, 2016-08-11:
I would of thought this issue would have been resolved by now, but on a PI3 using install  python2.7-dev, which solved my can't find python.h problem - thanks, I get this error. Help is needed please.

pi@raspberrypi:~/python $ sudo python PIstuffing.py
Traceback (most recent call last):
File "PIstuffing.py", line 3, in 
from RPIO import PWM
File "build/bdist.linux-armv7l/egg/RPIO/init.py", line 115, in 
File "build/bdist.linux-armv7l/egg/RPIO/GPIO.py", line 7, in 
File "build/bdist.linux-armv7l/egg/RPIO/GPIO.py", line 6, in __bootstrap
SystemError: This module can only be run on a Raspberry Pi!

joshuabaldwin, 2016-09-10:
I am running into the exact same issue, with my Pi 3.  Has anyone found a solution to this?

waynejacobsen, 2016-09-10:
Pi 3 is a raspberry pi. Do you mean a particular modle?

Sent from Wayne's iPad

joshuabaldwin, 2016-09-12:
Wayne, the Raspberry Pi 3 gets the error message generated by line 533 of source/c_gpio/py_gpio.c

`PyErr_SetString(PyExc_SystemError, "This module can only be run on a Raspberry Pi!");` 

I am running a Raspberry Pi 3 Model B.  The hardware changes are such that I cannot figure out how to replicate Cor Cornelisse's 9/10/2014 commit, which was designed to add support for the RPi 2B, to add support for the RPi 3B.

([github.com/metachris/RPIO/commit/fadf602816accaeba3f54a85b9f35a8181ec076c](https://github.com/metachris/RPIO/commit/fadf602816accaeba3f54a85b9f35a8181ec076c))

Yackou, 2016-09-12:
I have tried adding Pi3 support in my PR, in addition to Pi2 (for both GPIO and PWM). I have only validated on a Pi2 (don't have a Pi3), but I believe Pi3 support should be almost if not already working in my v2 branch.
Can you test it?
https://github.com/metachris/RPIO/pull/82
or 
https://github.com/Yackou/RPIO/tree/v2

waynejacobsen, 2016-09-12:
I would be happy to but, understand I'm old and slow. I could send you my pi3 on the basis that you will return it when finished. I'll pay round trip postage.

   

https://github.com/Yackou/RPIO/tree/v2—

joshuabaldwin, 2016-09-12:
I'm trying to figure out how to clone your installation on my Pi3 for testing. Attempting this:

`git clone https://github.com/Yackou/RPIO/tree/v2/RPIO.git`

is giving me a fatal repository not found error.

waynejacobsen, 2016-09-13:
Ill try v2 tomorrow on pi3

Sent from Wayne's iPad

Yackou, 2016-09-13:
@joshuabaldwin:
The following commands should work:
`git clone https://github.com/Yackou/RPIO.git`
`cd RPIO`
`git checkout v2`

surak, 2016-10-26:
1) What about pip?
2) Do we always need root for this?

VSkublin, 2017-04-10:
I have faced the same problem, looked for solution and this is the solution, that make it works:
First need to be sure that header files and static libraries for python dev. installed  properly:
sudo apt-get install python-dev  # for python2.x installs
sudo apt-get install python3-dev  # for python3.x installs
Then try to install RPIO branch v2, it will work with RPI3.
cd ~
git clone https://github.com/metachris/RPIO.git --branch v2 --single-branch
cd RPIO
sudo python setup.py install

This solution helped me install RPIO for several RPI3s.

TafThorne, 2017-07-21:
This looks like a duplicate of #78

JohnPeel, 2017-10-13:
RPi3 in 64bit mode doesn't work.

I assume because /proc/cpuinfo has slightly changed to not have any information on what board you're using.

`rpi3 ~ # cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

rpi3 ~ #
`

Jezorko, 2018-12-18:
Neither @Yackou nor @VSkublin solution works for me.

Contents of `/proc/cpuinfo` on my Raspberry:
```
# cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```

I'm running Arch Linux ARM:
```
# uname -srvmo
Linux 4.16.0-1-ARCH #1 SMP Mon Apr 2 18:40:18 MDT 2018 aarch64 GNU/Linux
```

ebarreraz, 2019-04-07:
I had the same problem with Ubuntu arm. With raspbian everything works perfectly

blitzcaster, 2019-05-16:
Still no solution for Ubuntu 18.04 aarch64?

minamagdyshehata, 2019-06-29:
i have facing this same problem on RPi3
why can't we update /proc/cpuinfo and add the missing line ?!!

DevXXM, 2019-07-06:
> 我在RPi3上遇到同样的问题
> 为什么我们不能更新/ proc / cpuinfo并添加缺失的行？!!
> 
> 编辑:)
> 
> 大家好，我的问题已经_解决了_，现在我可以在我的RPi 3 B +上使用RPi.GPIO和kali 64位。
> 
> 使用以下命令复制/ proc / cpuinfo：
> 
> > sudo cp / proc / cpuinfo / root / cpuinfo_fake
> 
> 使用nano编辑cpuinfo_fake并添加缺失的行（硬件，修订版和序列号）
> 
> > nano / root / cpuinfo_fake
> 
> 这些是我添加的行：
> 
> > 硬件：BCM2837 
> > 修订：a02082 
> > 序列号：0000000000000000
> 
> 然后保存文件。
> 
> 现在使用此命令挂载cpuinfo_fake
> 
> > sudo mount --bind / root / cpuinfo_fake / proc / cpuinfo
> 
> 现在检查它是否有效：
> 
> > cat / proc / cpuinfo
> 
> 你可以使用这个命令取消伪造的文件：
> 
> > sudo umount / proc / cpuinfo
> 
> 每次你重新启动你的RPi cpuinfo文件将恢复正常>
> 
> 现在要使用python脚本，你将需要在你的脚本开头的那两行，然后正常使用其余的代码：
> 
> > import os 
> > os.system（“sudo mount --bind / root / cpuinfo_fake / proc / cpuinfo”）
> 
> 这整个方法在这里解释：[https](https://www.reddit.com/r/Crostini/comments/9g6e6y/matlab_issues_with_crostini_any_ideas/e69p8a5/)：
> [//www.reddit.com/r/Crostini/comments/9g6e6y/matlab_issues_with_crostini_any_ideas/e69p8a5/](https://www.reddit.com/r/Crostini/comments/9g6e6y/matlab_issues_with_crostini_any_ideas/e69p8a5/)

thank you !