metachris/RPIO #94: UnicodeDecodeError

pipeult, 2017-03-24:
Hello,

After the "SystemError: This module can only be run on a Raspberry Pi!" error I found the RPIO v2 which is sensible to solve the problem.
Now, I found the second:
sudo easy_install -U RPIO return "UnicodeDecodeError: 'utf-8' codec can't decode byte..."

My env is RPi3, python3.6. Any change to my locale return the same error.

pipeult, 2017-03-24:
To try to take out, I passed in python version 2.7 instead of 3.6 on my RPi3 (Minibian OS)
Here my command to the install:

`easy_install https://github.com/metachris/RPIO/archive/v2.zip`
`Downloading https://github.com/metachris/RPIO/archive/v2.zip`
`Processing v2.zip`
`Writing /tmp/easy_install-MDwQzI/RPIO-2/setup.cfg`
`Running RPIO-2/setup.py -q bdist_egg --dist-dir /tmp/easy_install-MDwQzI/RPIO-2/egg-dist-tmp-zO69FW`
`source/c_gpio/cpuinfo.c: In function ‘get_cpuinfo_revision’:`
`source/c_gpio/cpuinfo.c:50:9: warning: ignoring return value of ‘fgets’, declared with attribute warn_unused_result [-Wunused-result]`
`         fgets(buffer, sizeof(buffer) , fp);`
`         ^`
`In file included from source/c_pwm/pwm.c:88:0:`
`source/c_pwm/mailbox.h:34:1: warning: function declaration isn’t a prototype [-Wstrict-prototypes]`
` int mbox_open();`
` ^`
`In file included from source/c_pwm/mailbox.c:38:0:`
`source/c_pwm/mailbox.h:34:1: warning: function declaration isn’t a prototype [-Wstrict-prototypes]`
` int mbox_open();`
` ^`
`source/c_pwm/mailbox.c:246:5: warning: function declaration isn’t a prototype [-Wstrict-prototypes]`
` int mbox_open() {`
`     ^`
`In file included from source/c_pwm/pwm_py.c:27:0:`
`source/c_pwm/mailbox.h:34:1: warning: function declaration isn’t a prototype [-Wstrict-prototypes]`
` int mbox_open();`
` ^`
`zip_safe flag not set; analyzing archive contents...`
`Adding RPIO 2.0.0-beta1 to easy-install.pth file`
`Installing rpio script to /usr/local/bin`
`Installing rpio-curses script to /usr/local/bin`

`Installed /usr/local/lib/python2.7/dist-packages/RPIO-2.0.0_beta1-py2.7-linux-armv7l.egg`
`Processing dependencies for RPIO==2.0.0-beta1`
`Finished processing dependencies for RPIO==2.0.0-beta1`

I'm not C programmer and I don't known if it can be important:
Keywords returned by `nano /proc/cpuinfo` are separated from their value by tabulations and not from spaces. The line 51 `sscanf(buffer, "Hardware	: %s", hardware);` use spaces.

dalanmiller, 2019-08-23:
Same experience on Raspberry PI Zero W