metachris/RPIO #96: Support for Raspberry Pi Zero W

jjorosa, 2017-05-30:
Hi, 

I'm afraid that maybe the Model/Revision for the rpi zero w isn't being handled correctly. I get this when I run sudo rpio --sysinfo:

Traceback (most recent call last):
  File "/usr/local/bin/rpio", line 4, in <module>
    __import__('pkg_resources').run_script('RPIO==2.0.0-beta1', 'rpio')
  File "build/bdist.linux-armv7l/egg/pkg_resources.py", line 534, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "build/bdist.linux-armv7l/egg/pkg_resources.py", line 1445, in run_script
    exec(script_code, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/RPIO-2.0.0_beta1-py2.7-linux-armv6l.egg/EGG-INFO/scripts/rpio", line 353, in <module>
    
  File "/usr/local/lib/python2.7/dist-packages/RPIO-2.0.0_beta1-py2.7-linux-armv6l.egg/EGG-INFO/scripts/rpio", line 346, in main
    
  File "build/bdist.linux-armv6l/egg/RPIO/__init__.py", line 186, in sysinfo
KeyError: '9000c1'

Indeed, the 9000c1 is the revision number for the raspberry pi zero w.

I guess it's easy to fix it, but I just can't figure what to move without opening the gates of hell.

Thanks in advance.

JamesGKent, 2017-07-26:
because of this issue i ended up forking and patching so i could use on my own pi zero w
my fork was taken from a v2 branch which had some other changes, not sure which version you were working with, but you could give it a try and see if it works for you (and let me know anything that doesn't)
the fork is here: [https://github.com/JamesGKent/RPIO/tree/v2](https://github.com/JamesGKent/RPIO/tree/v2)

Dave4675, 2017-08-30:
@JamesGKent I installed your version but it says during installation, that there is some error with cpuinfo ends with the problem that I can't install your version, also I added my raspberry pi 0 W manually to the __init__.py to th v2 of metachris reading your because it is 9000c1, but now I don't know if is a different problem, but when I try the servo.py example it give to me .

Dave4675, 2017-08-31:
@JamesGKent  it work on my raspberry pi 0 w but, the installation was exiting with -1 because you have to add #include <stdint.h> to cpuinfo.c. Anyway, thank you very much for creating your version

VanBlood, 2017-09-08:
Hi, I'm having problems with my raspberry zero w and RPIO too, I tried James RPIO and I made the Dave modification. Thanks to both, I'm little more close.

RPIO --version now works, but when i try to import RPIO i get this message:

`>>> import RPIO
Traceback (most recent call last):
  File "<pyshell#1>", line 1, in <module>
    import RPIO
  File "/usr/local/lib/python3.4/dist-packages/RPIO-0.10.1-py3.4-linux-armv6l.egg/RPIO/__init__.py", line 115, in <module>
    import RPIO._GPIO as _GPIO
ImportError: No module named 'RPIO._GPIO'
>>> `

Dave4675, 2017-09-09:
did you installed it using "sudo python3 setup.py install" ?

VanBlood, 2017-09-11:
> Did you installed it using "sudo python3 setup.py install"?

Yes

> Type "groups" in your raspberry, if you don't find gpio then you should add it

It is

> Or try this "sudo apt-get -y install python-rpi.gpio"

Was already installed with the newest version

I opened the IDE thorny with sudo, and now it seems to read "import RPIO" but the execution stops in the first intruction "servo.set_servo(pin, pos)", I mean:

```rusthon
from RPIO import PWM
import time

start=1.3 #ms
inc=0.004
end=1.7
servo_pin=14

servo1=PWM.Servo()
servo1.set_servo(servo_pin, start*100)
print('Start')

for i in range(start,end,inc): #range((end-start)/inc):
    servo1.set_servo(servo_pin, i*inc+start)
    time.sleep(0.01)
    read_ultrasonic(trigger, echo)
    print ('cycle = ', i)
```
When I run this test, servo move to the start position, but i never see "Start" in console, seems the execution freeze in "set_servo"

Thanks for help me

Dave4675, 2017-09-11:
When I use the print command I put the words into double quotes, anyway analizing your code, you have a period of 50Hz as default, and you are giving him pin 14 with a pulse of 130. Actually this is strange because the standard pulse range of servos are from 1000 to 2000, you haven't to worry about that, in my program for strange reason this range has changed two times. Another issue that I found was the problem that because the ervo was requiring a certain current that cause a dangerous voltage drop that made my raspberry reboot, so then what happen to your rasperry when it down't run your program properly?

VanBlood, 2017-09-11:
Thanks I forget a 0 and *1000 below and another punch of things... wow that was a big noob misstakes hahaha, thanks a lot. I was blinded with the problems with RPIO. Now all seems to work well. Although I keep having to run RPIO with sudo.

PD: My raspberry it's fine I was using a led to test the pwm before use a servo :D

jjorosa, 2017-09-12:
Thanks to @JamesGKent, @Dave4675 and @VanBlood. Now is working like a charm!

blueslevin, 2017-11-03:
Hi Everybody!

Thank you very much for this version of RPIO, it seems to be the only one working with Raspberry PI Zero W. However, I am experiencing a critical issue with DMA and PWM. My raspberry completely freeze after a random amount of second that I execute the following script from thonny (runned as sudo):

`from RPIO import PWM

PWM.setup()
PWM.init_channel(0)
PWM.add_channel_pulse(0, 17, 0, 50)`

With "freeze" I mean completely, I am not able to move the cursor. Still, the PWM continues to run on the selected pin even after the freeze.

Do you have any idea about what could create such a problem?

DEvil0000, 2017-11-03:
As far as I remember there are some channels used by the OS and this may conflict. Not sure if this actually is true or the case here. Try using a different one.

JamesGKent, 2017-11-06:
i found a similar issue using channel zero, i switched to channel 1 and it works flawlessly, never figured out the reason but as using channel 1 worked i never got around to investigating too deeply.

blueslevin, 2017-11-06:
Thank you very much guys! I was using channel 0 as well. I'll give it a try with other channels.
Thanks again!