metachris/RPIO #98: Doesn't work on Linux 4.9.0 with aarch64

russell, 2017-07-24:
From what i can see some of the expected values from `/proc/cpuinfo` when running with a aarch64 kernel are missing.  This is causing the board detection to fail.  But it seems that there is some device info is also listed in `/proc/device-tree/model`  is there a possibility to add a fall back to that value for detection?  The image I'm using is listed on this page https://wiki.debian.org/RaspberryPi3

I've tested this with the 0.10.0 and the v2 branch.

```
/root/plantower/venv/src/rpio/source/RPIO/__init__.py in <module>()
    113 """
    114 from threading import Thread
--> 115 import RPIO._GPIO as _GPIO
    116 from RPIO._RPIO import Interruptor
    117 

SystemError: This module can only be run on a Raspberry Pi!
```

```
# uname -a
Linux rpi3 4.9.0-2-arm64 #1 SMP Debian 4.9.13-1 (2017-02-27) aarch64 GNU/Linux
```

```
# cat /proc/cpuinfo 
processor	: 0
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
```

```
$ cat /proc/device-tree/model
Raspberry Pi 3 Model B Rev 1.2
```

metachris, 2017-07-24:
Thanks for the detailled report. Will look into it.

JamesGKent, 2017-07-31:
can you check if this file exists: /proc/device-tree/soc/ranges