metachris/RPIO #104: BCM2835 support?

courageon, 2018-02-25:
Is there support for the BCM2835 based io? I've got a PI-B revision (one of the first) and I'm getting the "This module can only be run on a Raspberry Pi!" exception when I try to import the module. 
running "cat /proc/cpuinfo" gives these hardware specs for my Pi (if that is helpful):
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 0002
Serial          : 0000000024e08e00

When looking through the source for this project I noticed hard-coded values for the broadcom module "BCM2708". Are there forks for the 2835?

Thanks!

courageon, 2018-02-25:
Forked it, got it working...
https://github.com/courageon/RPIO

metachris, 2018-02-26:
This repository is open for someone else to take over maintenance. Interested?

DEvil0000, 2018-02-26:
I said it once again. I would like to do this and spend some time - this said I may not be the guy having time to have a close look within a week.I also have just a limited set of PIs: old 1 rev (0x3), new 1 revisions , b+, zero, 3

courageon, 2018-02-27:
I'd be down for taking some ownership. Like @DEvil0000 though, I too have a limited set of PIs, just some first gen Bs and a PI 3. But I really like the idea behind this library, super clever trick for getting an awesome PWM out of the PI.

metachris, 2018-02-27:
👍 

what do you think would be good next steps for improving this repo? i'm open to adding new people as maintainers and move the RPIO pypi project to a common space.

courageon, 2018-02-27:
Well, tbh I think trying to create a "common" base with  https://github.com/xinkaiwang/rpio-pwm would be useful (since that's how I got this one working again, at least on my PI). Consolidate the low level code and branch for the python and node versions. I can reach out and ask if that's something @xinkaiwang would like to do. Keeping in mind some of that code was also borrowed from other repos, so that spider-webs very quickly.

Past that I think just testing would be the next big steps, but I think that's more on the community since this is an open sourced effort. I've been looking for an excuse to get another PI3, so I can definitely take on some testing with that one :)

Yackou, 2018-02-27:
Sorry for pitching in late, but have you looked at @JamesGKent repo or mine, and at pull request #100 ?

It seems several people have been looking at RPIO and ended up fixing the same board and kernel support issues over and over, so committing those fixes would definitely be helpful. As I said some months earlier, I don't think I'd be up for maintaining the repo all by myself, but I'd certainly be willing to participate and help move things forward.

DEvil0000, 2018-02-28:
I did not have a close look at forks till now.

I think there are some steps needed next:
1) adding 2-5 guys to do get this back running
2) get low level code nice, clean and shiny
3) merge the improvements and fixes starging with the issues most people face
4) maybe split python wrapping to a second repo

I think its not too much work to get it mostly straight again. As soon as some people are on board we should have a quick discussion or meeting to figure out details.

JamesGKent, 2018-02-28:
i think the biggest thing to start with would be merging some fixes for the newer cpus. but ultimately i think we need to implement the following:

- interpret serial number according to rpi foundation guidance to determine base address for peripherals
- implement checking to ensure dma channel being used is not being used by other processes

and nice to haves:
- in the event that a board is not supported directly we could add functions to specify the base address/any other addresses manually
- simple pwm functions to control a pin pulsing at a certain frequency and duty cycle (start/stop/set duty/set freq etc)

DEvil0000, 2018-03-05:
@metachris so how do you want to proceed on this?

DEvil0000, 2018-03-07:
I think last time I was reading the code was at about 0.7.X
I was never a fan of the python coupling and build process. Somehow its nice but in some ways its coupled to close and to pythonish.
However I think you @metachris should point out whats the official difference between master and v2 branch and why there is a v2 branch. Also when you should use the v2 branch - or just eliminate the need of the v2 branch and merge back to master.
Beside this you should merge compatibility from PR #82 or others (maybe ealier ones) doing the same thing. (Or pick the best and merge the rest)
I would like to help and invest some of my time but you did not give me access till now and I do not want to see the work ending up in another PR which is never getting merged.

DEvil0000, 2018-03-14:
@metachris you asked now the third time for help/someone to take over but everytime you ignore all replies. Here are some people willing to help out so why do you ignore it/them and ask again one year later?
Thats quite annoying...