metachris/RPIO #106: This module can only be run on a Raspberry Pi

darrahts, 2018-03-05:
Getting this error.  Installed with the following commands: 

git clone <url>
cd RPIO
sudo python3 setup.py install
python3
>>>import RPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/RPIO-0.10.1-py3.4-linux-armv7l.egg/RPIO/__init__.py", line 115, in <module>
    import RPIO._GPIO as _GPIO
SystemError: This module can only be run on a Raspberry Pi!

darrahts, 2018-03-05:
running rpio-curses also gives the same error

DEvil0000, 2018-03-05:
Which PI version do you use?
What is the exact hardware version?

Try one of the other branches / pull requests

darrahts, 2018-03-05:
from cat /proc/cpuinfo

Hardware : BCM2709 ----- I thought pi's used BCM2835?? 
Revision : a22082 ----- (rev 3 board)

I tried this fork https://github.com/JamesGKent/RPIO that says it works for all pi's but it gave the same error

DEvil0000, 2018-03-05:
As far as I know the different PI versions use BCM2835/2836/2837.
However the first google hit tells us it is reported incorrect when running a 32bit OS.
https://www.raspberrypi.org/forums/viewtopic.php?t=140572

As you can lookup here bx revision its a PI 3 model:
https://elinux.org/RPi_HardwareHistory

So guess there are multiple things you can do:
a) use a 64bit OS version
b) modify the code to detect this one same as the corresponding pi3 version
c) wait until someone fixed detection in 32bit OS or someone did write a workaround in this lib

darrahts, 2018-03-05:
I'm running rasbian jessie. There are others with this problem and I havn't seen a workable solution, ive done a lot of searching on this and sort of coming in circles. 
In the link @DEvil0000 sent, the second post explains that all pi distros are 32 bit, so if that is the case, how can a library written for the pi not work on 32 bit?

darrahts, 2018-03-05:
thank you for your help btw

DEvil0000, 2018-03-05:
Guess it all comes down to model detection again. Will have a look with my pi 3 at the weekend.

DEvil0000, 2018-03-07:
Today I had some time to investigate even before the weekend.
My PI 3 Board is revision a02082 as it seems.
Running latest raspbian image (9.1 stretch) with kernel:
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux

First of all - /proc/cpuinfo information:
```
Hardware	: BCM2835
Revision	: a02082
```
so the SOC is not detected correct but "better" then on yours. The revision is possibly correct.
This means we can not trust or depend the hardware detection but the revision.

As far as I can say on the master branch here only supports PI 1. So cloning master does not help v2 branch is better but even then it does not support PI zero and PI 3 models.
As you can read in the PR #82 the fixes are also in the v2 branch on https://github.com/JamesGKent/RPIO

This works for me on my board (only tested the basic functionality)

Wikinaut, 2018-06-11:
Same problem here with Raspberry 2 B V1.1.

Wikinaut, 2018-06-11:
```cat /proc/cpuinfo

processor       : 0
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 1
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 2
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

processor       : 3
model name      : ARMv7 Processor rev 5 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xc07
CPU revision    : 5

Hardware        : BCM2835
Revision        : a01041
Serial          : 00000000c8f36152
```

PrzemoF, 2018-09-24:
Same here on the pip3 install:
``pi@raspberrypi:~/Open-Cycling-Computer $ uname -a
Linux raspberrypi 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux
pi@raspberrypi:~/Open-Cycling-Computer $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 000000005cba30cc
pi@raspberrypi:~/Open-Cycling-Computer $ python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/RPIO/__init__.py", line 115, in <module>
    import RPIO._GPIO as _GPIO
SystemError: This module can only be run on a Raspberry Pi!
>>> 
pi@raspberrypi:~/Open-Cycling-Computer $ pip3 search RPIO
[..]
RPIO                      - Advanced GPIO for the Raspberry Pi. Extends RPi.GPIO with PWM, GPIO interrups, TCP socket interrupts, command line tools and more
  INSTALLED: 0.10.0 (latest)

Branch v2 works fine with sudo:
``pi@raspberrypi:~/RPIO $ python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.4/dist-packages/RPIO-2.0.0_beta1-py3.4-linux-armv6l.egg/RPIO/__init__.py", line 115, in <module>
    import RPIO._GPIO as _GPIO
RuntimeError: No access to /dev/mem. Try running as root!
>>> 
pi@raspberrypi:~/RPIO $ sudo python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPIO

tim1mw, 2018-12-31:
Just tried installing RPIO using easy_install on a pi zero with raspian stretch and I'm getting the same error:

Traceback (most recent call last):
  File "servo3b.py", line 2, in <module>
    from RPIO import PWM
  File "/usr/local/lib/python3.5/dist-packages/RPIO-0.10.0-py3.5-linux-armv6l.egg/RPIO/__init__.py", line 115, in <module>
    import RPIO._GPIO as _GPIO
SystemError: This module can only be run on a Raspberry Pi!

Is there any way to simply bypass the hardware check when it fails?

pierrotsmnrd, 2020-09-27:
I had installed python3-rpi.gpio with pip. I realized the version was wrong. I uninstalled it, downloaded from the source ( https://sourceforge.net/p/raspberry-gpio-python/code/ci/default/tree/RPi/ ) and built and installed manually (`python3 setup.py build && python3 setup.py install`), and now it works - even on my Raspberry Pi 4 .

The last version has a better handling of the last models of raspberry pis and their revision codes ( https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md ).

dalanmiller, 2020-10-06:
Just had this same issue happen for me on Raspberry Pi Zero

```
pi@raspberrypi:~/repos/RPIO $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 9000c1
Serial		: 00000000873eeabf
Model		: Raspberry Pi Zero W Rev 1.1
```

pierrotsmnrd, 2020-10-06:
@dalanmiller have you tried to install the `python3-rpi.gpio` package from source, as I mention in my previous comment ?