metachris/RPIO #111: SystemError: This module can only be run on a Raspberry Pi!

hosnerm, 2019-03-05:
Hi everyone 

i tried to run the example3_pwm_servo.py on my Raspberry 3B+ and it gave me the following error when trying to launch the python file : 

   `Traceback (most recent call last):
     File "example3_pwm_servo.py", line 1, in <module>
       from RPIO import PWM
     File "/home/pi/.local/lib/python2.7/site-packages/RPIO/__init__.py", line 115, in <module>
       import RPIO._GPIO as _GPIO
   SystemError: This module can only be run on a Raspberry Pi!`

Could it be that this has to do with using the Rpi 3B+?  
I looked through issues #78 #53 #86 and #106 and tried the solutions that were proposed and verified in there but they did not help. 

Any help would be greatly appreciated!

hosnerm, 2019-03-25:
by using the pigpio library from here http://abyz.me.uk/rpi/pigpio/index.html i got it to work without any jitter on the servo, think this makes quite a good alternative for all who are struggling with this library.

johnroyer, 2019-07-03:
I use RPIO version 2. This problem is still exists.

```bash
wget https://github.com/metachris/RPIO/archive/v2.zip
```