metachris/RPIO #117: RuntimeError: This module can only be run on a Raspberry Pi!

aceboy2020, 2019-11-14:
I am using Kali linux arm in my raspberry pi 4. I install crowpi drivers because for me it is useful to learn in complete setup kit. After i execute example files in crowpi this error shows.

root@kali:~/CrowPi/Examples# sudo python buzzer.py
Traceback (most recent call last):
  File "buzzer.py", line 5, in <module>
    import RPi.GPIO as GPIO
  File "/usr/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!

And also this..

root@kali:~/CrowPi/Examples# sudo python lcd.py
Traceback (most recent call last):
  File "lcd.py", line 6, in <module>
    import Adafruit_CharLCD as LCD
  File "build/bdist.linux-aarch64/egg/Adafruit_CharLCD/__init__.py", line 1, in <module>
  File "build/bdist.linux-aarch64/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 98, in <module>
  File "build/bdist.linux-aarch64/egg/Adafruit_CharLCD/Adafruit_CharLCD.py", line 104, in Adafruit_CharLCD
  File "build/bdist.linux-aarch64/egg/Adafruit_GPIO/GPIO.py", line 429, in get_platform_gpio
RuntimeError: Could not determine platform.

aceboy2020, 2019-11-14:
and also this 

root@kali:~/CrowPi/Examples# sudo python matrix_demo.py
Traceback (most recent call last):
  File "matrix_demo.py", line 40, in <module>
    main(cascaded=1, block_orientation=90, rotate=0)
  File "matrix_demo.py", line 21, in main
    serial = spi(port=0, device=1, gpio=noop())
  File "/usr/local/lib/python2.7/dist-packages/luma.core-1.12.0-py2.7.egg/luma/core/interface/serial.py", line 284, in __init__
    raise luma.core.error.DeviceNotFoundError('SPI device not found')
luma.core.error.DeviceNotFoundError: SPI device not found

 With the command lsmod
I can see (vc_sm_cma              36864  1 bcm2835_mmal_vchiq)

Is this impossible for me to use Kali as an OS to execute crowpi modules/drivers?
I bought crowpi because it has complete kit.. Such as matrix,lcd,breadboard,sensors,led and can be attach to raspberry 4.
This is crowpi https://www.elecrow.com/pub/media/wysiwyg/new_products/crowpi_diagram_1.jpg

joan2937, 2019-11-14:
@aceboy2020 This is the repository for RPIO.  You seem to be using RPi.GPIO.