metachris/RPIO #125: Problem running RPIO on Raspberry Pi 4

Tim8Trudeau, 2021-06-16:
I am seeing a "SystemError: This module can only be run on a Raspberry Pi!" I am running RPIO on a Raspberry Pi 4 under Python 3.7.3. 
Logged into the Pi via SSH from a windows 10 machine.

Here is the traceback from import RPIO. 

```
pi@raspberrypi:~ $ sudo rpio --version
Traceback (most recent call last):
  File "/usr/local/bin/rpio", line 345, in <module>
    main()
  File "/usr/local/bin/rpio", line 147, in main
    import RPIO
  File "/usr/local/lib/python3.7/dist-packages/RPIO/__init__.py", line 115, in <module>
    import RPIO._GPIO as _GPIO
SystemError: This module can only be run on a Raspberry Pi!
```

The OS is buster

```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

```
Is there a work around? Please respond to tim.trudeau@assaabloy.com

kevinvugts, 2021-07-30:
Hi @Tim8Trudeau,

Were you able to find a work around?
I am having the same issue.

The author @metachris  of this library doesn't seem to be very responsive

Tim8Trudeau, 2021-08-04:
I dropped RPIO in favor of RPi.GPIO. 

From: Kevin Vugts
Sent: Friday, July 30, 2021 11:46 AM
To: metachris/RPIO
Cc: Tim Trudeau; Mention
Subject: Re: [metachris/RPIO] Problem running RPIO on Raspberry Pi 4 (#125)

Hi @Tim8Trudeau,
Were you able to find a work around?
I am having the same issue.
The author @metachris of this library doesn't seem to be very responsive